Akaflieg-Freiburg/enroute #471: App Crashes Due to Unexpected Undo Gesture on iOS Devices

schneisi, 2024-11-12:
**Describe the bug**
When using the Undo gesture, the App crashes because it is not handled.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Aircraft Page
2. Swipe left with 3 fingers
3. Crash
It usually requires multiple attempts to get it right

**Smartphone (please complete the following information):**
 - Device: iPhone 14
 - OS: iOS 18.1

schneisi, 2024-11-13:
I’m postponing further investigation of this issue until after the migration to Qt 6.8. The problem only appears on the Aircraft page and doesn’t occur frequently in production.
Given that we’ll likely need to make QML adjustments as part of the migration, I’ll take a closer look if the issue persists in Qt 6.8.
For now, this can remain as is, with the expectation that the migration may resolve it.

kebekus, 2026-10-18:
Thank you very much, @schneisi, for this detailed report and for the clear steps to reproduce it! As far as I can tell, the problem no longer occurs since the migration to newer Qt versions, so I am closing this issue now.

Your reports really help to make Enroute better, so please keep them coming. If the crash with the undo gesture shows up again, just re-open this issue and we will take another look.

closed 2026-10-18 by kebekus