Akaflieg-Freiburg/enroute #473: Metar not showing

jetman104, 2024-11-17:
A METAR report is not showing on some airports even thou there a valid METARS existing. Exampe EDDL und EDLW. 
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jetman104, 2024-11-18:
Thanks for your quick answer, Stefan. A possible solution could be to leave everything as  is, but put a METAR link on the airport Page, which will be fed according  to your present  alghorism, and, when clicked and no data available yet, the METAR report will be uploaded. Thank you very much for your effort,  I love ENROUTE NAVIGATION!

charlylima, 2024-11-19:
Sometimes it happens that, after you add an airport to the Route, the METAR is already shown on the Weather view, but not yet in the Airport view. Maybe some update mechanism in the code is missing. 

Example:
Add EDDH Hamburg to your Route.
Look at the Weather view. METAR is shown.
Look at the Airport view. METAR is not shown.

kebekus, 2024-12-25:
Dear all, I will work on this issue next. This will take a while because it involves some non-trivial changes to our infrastructure and the METAR-providing code.

As a start, I wrote a script that generates a list of the ~10,000 known METAR stations worldwide, puts it up for download on our server, and updates it weekly. The app can later use the list to avoid asking for METAR from airfields without METAR stations.

kebekus, 2025-01-20:
Dear all, the problem should be solved in Program version 2.32.6, which appeared for Beta Testing on Google Play today. I had to re-write large parts of the METAR-handling code, with the potential to introduce new errors. Accordingly, I plan for a longer-than-usual beta testing phase. 

I appreciate your feedback!

charlylima, 2025-01-27:
Hello Stefan, 
Great that you are working on it. I tried it, it seems to always work when clicking at airfields in the map, sometimes with a little delay, i assume when it is downloading the data, that is perfect. 

I wonder, when i plan a route across germany and go to the weather view, there are some airfields in white color. Is it possible that they only have TAF data, but no METAR? For example ETSL, ETSN, LOXA, LSZL show only TAF. We should have a closer look at the data. Maybe they are closed at night and upload no METAR data? But i can find METAR data from these airports from 38 minutes ago at other internet services. 

The weather view should show the weather along the planned route?
I planned from EDPC in the south of germany to EDXR in the north. In the weather view it shows in between airports from switzerland and italy, which does not seem to make sense. For example LIPO, which is 315 km away from my route.

kebekus, 2025-02-05:
I apologize for the silence -- my job took the better of me. 

To answer @charlylima 

- There are stations with TAF but no METAR. This happens most often at airfields that produce METARs only when they are active, such as EDTL in Germany. Once the airport closes, the last METAR expires after 30 min and only the TAF remain -- note that TAFs are typically valid for a day or so.
 - Originally, we requested TAF/METAR data from the aviation weather center, for a region around the current position and the current route. We needed to change that for a number of reasons, and now we can only download data for a RECTANGULAR region. Since requests are relatively expensive and since the amount of data transferred in each request is relatively low, we have expanded the range to reduce the number of requests that our server needs to handle.

kebekus, 2025-02-05:
I believe that the issue is solved in version 2.32.8, which became available on Google Play today.

@jetman104 Thank you for pointing this out! Please feel free to re-open this request if necessary. Please keep reporting any issue that you may find in the future.

closed 2025-02-05 by kebekus