Akaflieg-Freiburg/enroute #474: Offer official ICAO maps, at least for Switzerland

kebekus, 2024-11-19:
The swiss official ICAO maps are free to use. We could offer them to our users.

tue-kyndal, 2024-12-06:
So are the danish ICAO maps.
https://aim.naviair.dk/kort/

I have made a few mbtiles my self.

The are here free to use for all.
http://tue.kyndal.dk/kort/

Have a look.
The Copenhagen area has the cites around Roskilde Airport with heavy noise restrictions and the aerobatic zones A-K!

I have merged 2 danish maps together for longer x-country flights. It was the best solution before you could switch maps..thanks for that feature ;O)

I have geo-refrenced the Billund area, and the flight paths for aero-towplanes around the glider club at EKGL.
Have fun with it.

kebekus, 2025-03-07:
Starting to work on this issue now.

kebekus, 2025-03-10:
The maps are now provided.

kebekus, 2025-03-10:
@tue-kyndal Thanks for pointing me to the maps for Denmark. I'll be more than glad to add them, but I need to be 100% sure that the license conditions allow me to do that. Perhaps it would be easiest if you could send me an email so we can discuss the matter in private? -- Thanks!

closed 2025-03-10 by kebekus