Akaflieg-Freiburg/enroute #476: WayPoint  Lib - add Tag; enable and disable within the map

joachimwe, 2024-11-24:
For Flightplanning its helpful to use waypoints (on lower enroute charts, eg. https://opennav.com/waypoint/IT) for crossboarder flightplans. Displaying all the information (waypoint) is too much when flying. If you could activate and deactivate a certain category of waypoints/library for the map would be perfect.

A user should define a lib-name and could add his own waypoints. He also can define, if the lib is shown on the map or in search dialogs. Selected waypoint should also be shown if they are selected in the route-plan, even if displaying is deactivated.
OR
A user could add a tag to a waypoint when importing and select the shown libs via tags (multi-selection should be possible)

Additional: If its possible to add the waypoint depending to a country via your importing mechanism would be sensational :-)
Waypoints could be extracted via https://opennav.com/waypoint/IT, https://opennav.com/waypoint/DE

TTC70, 2025-01-18:
Just a notification.
The list of waypoints in this link are IFR waypoints: https://opennav.com/waypoint/IT

I believe you know that Enroute is for VFR flight planning.

kebekus, 2025-01-19:
Dear all, I can see the use of this suggestion. However, I cannot promise that we will implement a solution quickly. We are an extremely small team; unless more people join the development, we have to prioritise strictly.