Akaflieg-Freiburg/enroute #477: EFN on Raspberry Pi 4

ocean71, 2024-11-24:
I have tried a couple of methods to run EFN on a Pi4.

To avoid compiling Stefan recommended using the available flatpak. Sadly that downloads OK but does not load on the Pi.

As far as the compiling alternative goes, the cmake .. command  shows problems straight away:

  Make Error at CMakeLists.txt:38 (find_package):
  Could not find a configuration file for package "Qt6" that is compatible
  with requested version "6.8".

  The following configuration files were considered but not accepted:

    /usr/lib/aarch64-linux-gnu/cmake/Qt6/Qt6Config.cmake, version: 6.4.2
    /lib/aarch64-linux-gnu/cmake/Qt6/Qt6Config.cmake, version: 6.4.2
    -- Configuring incomplete, errors occurred!

Clearly the problem lies with the Pi having an older version of Qt6 or some of its modules.
I tried altering the cmake.list file by changing 6.8 to 6.4 but it still failed. 

Stefan was asking how EFN ran on the Pi. Before EFN was upgraded to Qt6, I was able to compile each new version and it ran pretty well on the Pi, but since the adoption of qt6 I have run into compilation problems.

kebekus, 2024-11-26:
@hjr Helmut, here is a user who has problems compiling Enroute on a Raspberry Pi. Have you ever tried doing that?

kebekus, 2024-11-26:
@ocean71 It is indeed true that EFN required Qt 6.8. We do make good use of the new features in this release.

kebekus, 2024-12-01:
@ocean71 I am closing this issue report now because I have no solution for your problem. I regret that I have no better news and ask for your understanding: we are an extremely small team with very limited ressources, so that we need to prioritise strictly.

closed 2024-12-01 by kebekus

ocean71, 2024-12-01:
Thanks very much for trying. I quite understand. All the best