Akaflieg-Freiburg/enroute #480: GUI MAP: Zoom in Zoom out Buttons

bpmulder, 2024-12-21:
EFN V2.32.2
- Sugestion GUI zoom in and zoom out buttons: 
Down right hand corner there are two buttons for zooming in (plus) and zooming out (minus).
**Is it a consideration to give the plus button a fixed position** eg tapping the minus button several times at the end it is replaced by the plus button while the minus button has disappeared. 
Tapping the plus button several times to the max possible taps is not an issue because the plus button disapears and the minus button stays foot.

- Bonus, a small typo under menu (hamburger button), Settings, Primary Position Data Source, text last option Trafic data reveiver: receiver

kebekus, 2024-12-21:
@bpmulder Thank you for your suggestion! We will look at this and come back to you within the next few weeks.

kebekus, 2024-12-23:
All suggestions are implemented in Version 2.32.3, which is now available for Beta-Testing on Google Play.

bpmulder, 2024-12-23:
Version 2.32.3: Excellent! The zoom buttons stays on the same place. 
I like the implementation of the plus button very much, this is at it's max value the plus button is dimmed out. Perfect.
Because I like it so much I suggest to do the same for the minus button, this is at it's min value the minus button will also be dimmed out and still will be visble.
Great and useful improvement regarding the user interface due to the feedback it gives to the user. Thank you so much!

bpmulder, 2024-12-23:
Disregard my last suggestion with the minus button because you already did implement it! I did some further testing with other map backgrounds and saw the minus button very faint at some point. Thanks again.

kebekus, 2024-12-24:
Closing this issue because the improvements have appeared in the production version on Google Play.

@bpmulder Thank you so much for your contribution. Please keep reporting issues that you may find.

closed 2024-12-24 by kebekus