Akaflieg-Freiburg/enroute #487: Adding new waypoint, menu does not go away.

TTC70, 2025-01-25:
**Describe the bug**
The title says it. After the last update the menu, where you can choose what to do with that specific waypoint, does not go away automatically.
The use has to press "close" to go back to the map.

**To Reproduce**
Steps to reproduce the behavior:
1. Keep Pressed on the waypoint on the map (or double tap).
2. The window related to that waypoint appears.
3. Click on "route"
4. Click "add".
5. The window does not close automatically, it has to be closed by user.

**Expected behavior**
After clicking "add", "direct to", "insert" or "delete" the window should close automatically. 
This happens only from the last update. In the previous versions the window was closing right after clicking "add", "direct to", "insert" or "delete" .

**Screenshots**
N/a

**Desktop (please complete the following information):**
 N/a

**Smartphone (please complete the following information):**
 - tablet and Samsung galaxy s10 phone
 - latest o.s. on both devices.

**Additional context**
N/a

kebekus, 2025-02-04:
@TTC70 Thanks for the suggestion. Implementing this should be easy, so I put it high on my priority list.

TTC70, 2025-02-04:
Thanks Stefan.
More than an "implementation" is a "go back to the previous behavior".
Before the last update everything was working properly.
After one of the last updates the window of the waypoint stays there.

kebekus, 2025-02-05:
@TTC70 Thanks for the report. The issue is fixed in version 2.32.8, available for beta-testers on Google Play. The version will go into production on Google Play later today.

closed 2025-02-05 by kebekus

TTC70, 2025-02-05:
I confirm that the waypoint window goes away after clicking the behavior for that point.
This in the last update.
Thanks Stefan.