Akaflieg-Freiburg/enroute #488: Library does not open

peczynska, 2025-01-27:
I can't access my library, neither by selecting it from the menu nor while selecting the route. 

 - Device: Google Pixel 7
 - OS: Android 15
 - App Version: 2.32.6
 
I have also encountered another problem, that has been already reported - when I want to add a waypoint manually, the menu does not go away thus I can't select and add it.

TTC70, 2025-01-27:
I quote the issue.
It seems something happened on the last update.

kebekus, 2025-01-27:
@peczynska @TTC70 Dear all, thank you for reporting this issue. 

I have received similar reports on other channels as well and will start to investigate the issue now with priority #1. I apologize for the silence -- my job kept me away from Enroute for a couple of days. I will report back on this site as soon as I have any clue as to what's going on.

Best,

Stefan.

kebekus, 2025-01-31:
Thanks to everybody for reporting this issue. The problem is fixed in version 2.32.7, which is now available at Google Play.

closed 2025-01-31 by kebekus

TTC70, 2025-01-31:
Hey Stefan.
The version 2.32.7 is the current version I have installed as beta on my device, and it is the version where I found the issue.
At the moment there are no updated versions on GPlay.
Cheers.