Akaflieg-Freiburg/enroute #489: Compatability with Google maps

tue-kyndal, 2025-01-27:
It would be great if a shared location from Google maps, could be opened with enroute navigation, and automatically create a direct to flight plan.

There are many use cases.

Just to mention a few.

If flying with a guest, to go and see thir house from above.. It would be super simple to add the point to navigation. 

I tow gliders, and sometimes they land on a field. If possible we/I will fly over and tow them back. 

Instead of stressing with coordinate and LAT LON formats, a Google maps pin is all what would be needed.

Just an idea that occurred to me last summer..in just such a case. 

[image]

tue-kyndal, 2025-02-02:
Thanks.
I obviously tested it before making this request.

If I do as surgested and share a location from Google maps with enroute.. Nothing happens?

Latest googled map and enroute.
one plus 12 with latest OS oxygen 15

tue-kyndal, 2025-02-02:
Was just playing a bit with it.
Its the temporary redirect to a browser that doesn't work.

But, a workaround.. From Google maps app... Share with a browser... Then share again with enroute from that browser .. And wola.. It works.

I Still think this is a bug/issue 🤔😎

kebekus, 2025-03-23:
@tue-kyndal I can confirm the issue -- Google has changed their policies. I will check if the connection to Google Maps is still possible.

kebekus, 2025-03-25:
I had a deeper look at the problem. It issue is that Google has changed their policies and actively hides coordinates in the data that they share. The new policy is not applied uniformly, and results differ between apps and countries.

It seems that the only clean solution is the remove support for Google Maps altogether. I will do that with the next version update.

tue-kyndal, 2025-03-30:
It still works if you send Google maps coordinates from a browser. It's quite convenient for ad hoc locations... Picking up out landed gliders... Or Just ad hoc destination with guest who wants to see the farm.
Perhaps keep the option to add from a Google maps browser.

kebekus, 2025-04-13:
@tue-kyndal As you have probably seen, I remove the temporary "redirection browser" that no longer worked with Google Play. Location sharing still works if Enroute is able to extract coordinates from a location URL.

I am closing this now.

closed 2025-04-13 by kebekus