Akaflieg-Freiburg/enroute #490: The list of saved aircraft does not show up.

stefanrasmusen, 2025-02-02:
**Describe the bug**

The window/list to load a saved aircraft, does not show up.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to main menu select 'Aircraft' then click the 3 dots up to the right, then select 'View library'.
2. The error is that no list of previously saved aircrafts is shown.

**Expected behavior**
A list of saved aircrafts should pop up.

**Screenshots**

**Smartphone (please complete the following information):**
 - Device: [Samsung S24
 - OS: Android 14

[image]

 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
If I instead select Save to library, I can actually see the list of the 8 aircrafts I have saved previously.

kebekus, 2025-02-04:
@stefanrasmusen Thanks for reporting this issue. The problem *should have been* fixed in version 2.32.7, which appeared on Google Play a few days ago. Can you please check that (or if) you are using this latest version?

Looking forward to hearing from you -- Stefan.

stefanrasmusen, 2025-02-04:
Hello, Yes. This is solved in the latest version. Sorry to bother you.

Best Regards
Stefan

closed 2025-02-05 by kebekus