Akaflieg-Freiburg/enroute #491: Issue with importing flight plans from and to Google drive

TTC70, 2025-02-05:
**Describe the bug**
"Import" and "share" for flight plan does not work. A shared flight plan (GPX format) saved into Google drive, is not read correctly by EFN when the user tries to retrieve with the "import" function giving the "Error importing the file" - "Error reading the file".

**To Reproduce**
Steps to reproduce the behavior:
1. Save flight plan on Google drive.
2. Import a flight plan
3. Choose Google drive as source.
4. See error

**Expected behavior**
The flight plan saved on Google drive should be read by EFN.

**Screenshots**
N.A.

**Desktop (please complete the following information):**
 N.A.

**Smartphone (please complete the following information):**
 - Device1 tablet with the last version of android.
 - device2: Samsung galaxy s10 with the last version of android.
 

**Additional context**
Downloading the flight plan locally on the device, the import function seems working and the flight plan is read correctly.
It seems the issue is only on files read directly from Google drive.
An idea: the app could manage only .FPL and/or .pln format for flight plans.

kebekus, 2025-02-22:
At the moment, code for reading content URLs is simply not there. I will investigate if it can be added with reasonable amount of work. If we choose not to implement that, the user should at at least see a descriptive error message that explains possible workarounds.

kebekus, 2025-02-22:
On further investigation, it seems that we are affected by [this error in the Qt library](https://bugreports.qt.io/browse/QTBUG-118154).

kebekus, 2025-03-03:
I am currently implementing a fix that restores the expected behavior. I hope to have a beta out within a few days.

TTC70, 2025-03-03:
Stefan, if it helps. 
It seems that the flight plan, if saved in local, is correctly read by EFN.
Maybe the issue was related only when EFN reads directly from Google drive.
I don't know if this may help.

kebekus, 2025-03-04:
@TTC70 The issue should be fixed in v2.32.14, which just appeared on Google Play for beta-testers. Can you please check and confirm that the issue is solved for you?

TTC70, 2025-03-04:
Hey Stefan,
Everything is ok now. The app correctly imports a flight plan from Google drive, and the flight plan is correctly shown.
After another check, I imported different flight plans from any device to the Drive and vice versa, all flight plans are correctly exported to Drive and imported from Drive.
Happy to help.

kebekus, 2025-03-05:
@TTC70 Thanks for the feedback. Version 2.32.14 will go into production today.

closed 2025-03-05 by kebekus