Akaflieg-Freiburg/enroute #492: Toggle Airport chart from Airport popup

krheinwald, 2025-02-11:
It would be nice to have a button in the airport pop-up to toggle the airport chart on-the-fly and to not have to go through the menu.

kebekus, 2025-02-16:
@krheinwald Thank you for your suggestion. I will discuss that in the development team. Expectation management: we are a very small team, so it might take a while before we come back to you.

kebekus, 2025-02-17:
Implemented. Works well, but needs some fine tuning.

TTC70, 2025-02-18:
Hey Stefan,
I have just downloaded the last version of EFN, how does this implementation work?

krheinwald, 2025-02-18:
Same here. I see no button in 2.32.9 on Android.

kebekus, 2025-02-19:
@krheinwald @TTC70 The feature is available in the latest developer sources, but you must compile the app yourself to access it. This is a non-trivial task. I will announce once the feature appears for beta testing on Google Play. Sorry for the misunderstanding.

kebekus, 2025-02-21:
@krheinwald @TTC70 Enroute Version 2.32.10 is available for testing at Google Play now. Please have a look. I will close this issue, but feel free to re-open if things do not work for you.

Thank you for your help.

closed 2025-02-21 by kebekus

TTC70, 2025-02-21:
> [@krheinwald](https://github.com/krheinwald) [@TTC70](https://github.com/TTC70) Enroute Version 2.32.10 is available for testing at Google Play now. Please have a look. I will close this issue, but feel free to re-open if things do not work for you.
> 
> Thank you for your help.

Hey Stefan.
What am I suppose to see in the airport menu?
There is nothing new in any class of the airport.

krheinwald, 2025-02-21:
> What am I suppose to see in the airport menu?

[image]

TTC70, 2025-02-21:
> What am I suppose to see in the airport menu?
>
> image.png (view on web)

I don't see those buttons...
Do you need the airports plates installed or the app takes them from
somewhere?

krheinwald, 2025-02-21:
> Do you need the airports plates installed

Yes.

kebekus, 2025-02-22:
@TTC70 @krheinwald The answer is correct. I cannot provide airport plates because there is no reliable and free source that I can use. For Germany, you can use the [AIP Browser DE](https://mpmediasoft.de/products/AIPBrowserDE/help/AIPBrowserDE.html) to get a private copy of all airport plates. Other countries do not offer them at all, or in complicated formats, or with restrictive license conditions that make them unusable in practice.

TTC70, 2025-02-22:
> > Do you need the airports plates installed
> 
> I would guess yes. I have them installed.
> 
> [image]
> 
> https://akaflieg-freiburg.github.io/enrouteManual/03-tutorialAdvanced/04-vac.html

Could you help with the procedure to make an airport plate working for Enroute?
What software do you use? What application to make the images geo-tagged?
Every help is appreciated (I suppose this can help also the whole community).
Thanks.

krheinwald, 2025-02-22:
Just follow the link on the post you quoted ;)

kebekus, 2025-02-23:
@TTC70 There is a program "AIP Browser DE" that you can find on the internet. It produces airport plates for Germany, in the format supported by Enroute. You can try these. For Italy, I have no idea.