Akaflieg-Freiburg/enroute #493: Search bar for Approach Charts

polarjet, 2025-02-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

If i plan a flight route in Germany from Airport A to destination B, i would like to be able to look easily and fast  into the approach chart of airport B. All germany approach Charts are available /downloaded by me .
But at the moment I have to scroll long time downside for looking for my airport B in the approach Charts database

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I would like to be able to look easily and fast  into the approach chart of airport B. 
E.g a airport name search bar in the approach Charts database would be fine.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Maybe a solution would be that i get a list of all my dedicated choosen airports /waypoints of my flight route. E.g.
Route : A -B-C where B and C are airport waypoints, I want to be able to click on my route list a d the related approach chart is opening.

**Additional context**
Add any other context or screenshots about the feature request here.
Thank you for this good project !

kebekus, 2025-02-16:
@polarjet That sounds like a very good idea that we can and should quickly implement. Note that "quickly" is a relative term though: we are a very small team, and development is therefore slow.

kebekus, 2025-02-17:
Implemented. Needs to be tested before going into production.

polarjet, 2025-02-17:
Great, thank's a lot! 
Greetings
Martin

TTC70, 2025-02-18:
> Implemented. Needs to be tested before going into production.

Hey Stefan,
as said in another post, as beta tester I would like to test this implementation.
Do you think you could give more info about how to reach it and test it?
I was thinking something like "Hey beta tester, this implementation could be reached this this and this... could you check if it works properly?".
Cheers.

kebekus, 2025-02-19:
@TTC70 Understood, but we are not there yet. I will make an announcement once the feature appears for beta testing on Google Play. For now, you will need to compile the app yourself to access the feature.

kebekus, 2025-02-21:
@TTC70 @polarjet Enroute Version 2.32.10 is available for testing at Google Play now. Please have a look. I will close this issue, but feel free to re-open if things do not work for you.

Thank you for your help.

closed 2025-02-21 by kebekus