Akaflieg-Freiburg/enroute #494: TMZ not shown on map with RMZ (Glider) present

krheinwald, 2025-02-16:
There are several TMZ/RMZ (Glider) areas in northern Germany, e.g. EDXC, EDXN).

While the airspace pop-up correctly shows this, the map only shows 'RMZ'.

kebekus, 2025-03-23:
@krheinwald Thank you for the report. I checked the airspaces that you mention ... the issue is complicated. The airspace **are** drawn correctly, but because there are so many overlapping airspaces with identical boundaries, the map layout algorithm comes to its limits.

I am not sure we can find a good solution here -- and improvement that works for one place is bound to create issues elsewhere. The airspace situation is simply too messy. I will discuss with my colleagues, but I do not promise anything.

krheinwald, 2025-03-23:
Not knowing anything about the algorithm, my approach would be to draw the airspaces based on 'severity', so that TMZ is drawn over RMZ (Glider) and stays visible.