Akaflieg-Freiburg/enroute #495: App freezes when doing a long press on the map or calling some menu items (e.g. Points or Meteo)

Flying-Roger, 2025-02-18:
App freezes when doing a long press on the map or calling some menu items (e.g. Points or Meteo)

Steps to reproduce the behavior:
1. Launch the application
2. Go to the main menu
3. Click on Météo  (or Points)
=> App freezes

The application also freezes after a long press on the map (anywhere).
After 1 minute, the App disappears from the screen and it restarts when clicking on the app icon.

**Smartphone (please complete the following information):**
 - Device: Samsung A34]
 - Android 14 - OneUi 6.1
 - Enroute version 2.32.9

The app worked fine until recently, but the problem is now permanent, making Enroute unusable.
Clearing the cache did not solve the problem.

kebekus, 2025-02-19:
@Flying-Roger Thank you for the message. This is of course critical. I am a little lost because I cannot reproduce the issue on any of my devices -- this includes Samsung devices running on Android 14.

- Can you access the page "Information/About Enroute"? There, you could move to the tab "system" and access system log messages for Enroute. Is there anything suspicious? Could you perhaps provoke a crash, and then send us the system log, so we can have a look at what's going on?
- When the problems occur, is your phone connected to the internet? Can you access the following URL from you phone: https://enroute-data.akaflieg-freiburg.de/enrouteProxy/metar.php?format=xml&bbox=47.0089,6.35167,49.0089,9.34113
- In the system settings, have you modified any settings concerning "accessibility features" of your device?

Thanks for your help!

Best,

Stefan.

Flying-Roger, 2025-02-19:
System Log sent via direct email to Stefan.

kebekus, 2025-02-20:
@Flying-Roger Thank you for the data. I am, however, still completely puzzled. The log files do not show anything, and I cannot reproduce the issue on any of my devices or emulators. This makes debugging hard. I am currently a little clueless, but I will continue investigating.

Next round of questions:

- Do you have a second device where you could try if the problem also appears?
- Does a complete deletion/reinstallation of the app help?

Where are you based geographically? I am asking because debugging is typically much easier if I have access to the phone…

Best wishes,

Stefan.

Flying-Roger, 2025-02-20:
Good news: 
- deinstalling and reinstalling solved the problem.
- I successfully reinstalled my aircraft and waypoints.
However:
- installation of the maps/background/terrain was rather painfull, with France and Germany downloads stopping forever in the middle of the download (e.g. 50, 75% or 97%). It took many retries (stop/relaunch download again starting at 0%) to get them installed, but now they are.

kebekus, 2025-02-21:
@Flying-Roger Thank you for the feedback. I am closing this issue now because the issue seems resolved for you and, honestly, I have no idea what the problem was or what I could be doing.

Please do not hesitate at all to re-open this issue if problems show up again. Also, please keep reporting any problems that you may find. Thanks again.

closed 2025-02-21 by kebekus