Akaflieg-Freiburg/enroute #501: Invalid coordinate found when importing OpenAir Airspace

nallison, 2025-02-28:
**Describe the bug**
When importing an OpenAir airspace file on either Android or Linux Desktop the Import Airspace Data page issues warnings of "Invalid coordinate found".  I have been using a selection of OpenAir files including excerpts from the [OpenAir Definition](http://www.winpilot.com/UsersGuide/UserAirspace.asp) and all throw these warnings and do not import the airspace.  I have used airspace files that import correctly into XCSoar.

**To Reproduce**
Steps to reproduce the behavior on Linux:
1. Go to Map and Data Library
2. Click on Menu and Import...
3. Select an OpenAir_Test.txt file (attached)
4. See error

**Expected behavior**
I expect properly formatted coordinates to be accepted by the parser.

**Screenshots**
See attached PNG

[image]

[OpenAir-Test.txt](https://github.com/user-attachments/files/19032529/OpenAir-Test.txt)

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.6 LTS
 - Browser: Firefox
 - Version: Enroute Flight Navigation 2.32.12 via Flatpak

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S8
 - OS: Android
 - Browser: Firefox
 - Version: latest from Beta programme delivered by Play Store

**Additional context**
I could not find any information to advise if Enroute requires any particular coordinate lat and lon formatting and the precision of the DD:MM:SS.SSSS that is accepted in the OpenAir airspace file

kebekus, 2025-03-02:
@nallison Thank you for your report! I can reproduce the issue on my machine and will (hopefully) be able to fix the problem soon.

kebekus, 2025-03-02:
@nallison The issue is fixed in version 2.32.13, which appeared on Google Play today. Can you briefly confirm here that your problems are resolved?

I am closing this issue now, but feel free to re-open anytime when necessary. Please keep reporting any issue that you may find -- we need to the help of our users to provide the best software we can. 

Best wishes from Freiburg / Germany,

Stefan.

closed 2025-03-02 by kebekus

nallison, 2025-03-02:
@kebekus Confirming that on my Samsung Galaxy S8 OpenAir imports correctly now when I navigate to the file with file explorer and choose "Open with Enroute Flight Navigation".

When I tried to import from within Enroute Flight Navigation > Library > Maps and Data > Import and select the file, Enroute opens the form "Download in progress... Please stand by while we download the list of available maps from the server..."  and does not import the file.

Thank you for the code fix, inspecting the commit I am a little embarrassed I did not see the issue and suggest a patch.

Kind regards
Neil
Christchurch, NZ

kebekus, 2025-03-24:
@nallison Is the problem that you describe ("Download in progress....") reproducable with the latest version of the app? If so, this is something I should be looking at...

nallison, 2025-03-25:
> [@nallison](https://github.com/nallison) Is the problem that you describe ("Download in progress....") reproducable with the latest version of the app? If so, this is something I should be looking at...

I cannot reproduce this with v2.32.21 on my Samsung Galaxy S8.