Akaflieg-Freiburg/enroute #502: Further attempt to install EFN on Raspberry Pi 4

ocean71, 2025-03-01:
Not wanting to give up trying to get Enroute working with the Raspberry Pi 4, I decided to have a try at installing qt 6.8 on the Pi.

It involved a lot of installation time and used a large amount of space, but as I have an SSD I have plenty of room for the installation. So that was OK.

I managed to get CMake to work and complete. There was a warning about not finding PlatformGraphics that I did not understand, but the process seemed to finish OK, showing I had all the required libraries and dependencies.

I did have problems, however,  at first with the MapLibre Qt requirement. I therefore downloaded the binary for release 3.0 (the latest for Linux) even though the latest covers 6.7 and not 6.8. I unzipped it and left it in the Download foldre. I was not sure whether it should be installed and if so how.

I then put  the following Path instruction in the .bashrc so qt 6.8 earlier installed could  be located

filePATH=/opt/Qt/6.8:$PATH

 and  then ran Cmake in Enroute with the following command. 

cmake -DCMAKE_PREFIX_PATH=//home/xxxxxx/Downloads/maplibre-native-qt_v3.0.0_Qt6.5.3_Linux/lib64/cmake/QMapLibre ..

As mentioned above this seemed to work, except for the odd PlatformGraphics warning, which appeared in brown type not red and did not stop the Make process. 

However when it came to the Build process, frustratingly it completed to 98% and then failed!
The following message was in the console, and I am not sure how to solve the problem.

[ 98%] Linking CXX executable enroute
/usr/bin/ld: //home/xxxxx/Downloads/maplibre-native-qt_v3.0.0_Qt6.5.3_Linux/lib64/libQMapLibreLocation.so.3.0.0: error adding symbols: file in wrong format
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/enroute.dir/build.make:5064: src/enroute] Error 1
make[1]: *** [CMakeFiles/Makefile2:868: src/CMakeFiles/enroute.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

I think it may be related to the fact that the binary I downloaded for Maplibre was for Linux and not specifically for the Pi. Or it may be because the Maplibre was for 6.7 not 6.8.

I could try to build Maplibre qt from scratch, but I suspect that won't be an easy task and I would imagine the number of submodules is considerable. Also I am not sure that is the solution.

Anyway, would appreciate any comments/advice on the console message above.
It is such a shame to get so close  to installation!

Thank you very much.

ocean71, 2025-03-23:
I think it makes sense to close this now, as there has been no response. I posted a similar query on the maplibre qt site and had no response there either.
It looks like this is a definite no starter for a raspberry pi at the moment, so I am happy for this to be closed now and I will abandon trying to get it to work on a Pi.  A shame, but never mind. 
It is a great app by the way and I have it on my Android tablet, so that's fine.

markus2marks, 2025-05-08:
i have created a yocto build that creates an image that launches the enroute app. i will upload it to my repository in the next few days.

markus2marks, 2025-07-15:
https://github.com/markus2marks/meta-openvario/tree/enroute

kebekus, 2025-07-19:
@ocean71 @markus2marks Thank to both of you. If I understand correctly, then Enroute is now available on the Raspberry, so I am closing this issue now. Feel free to re-open if anything new comes up!

closed 2025-07-19 by kebekus