Akaflieg-Freiburg/enroute #503: Show info box outside of 3 NM Corridor

PhilippBuehler, 2025-03-03:
I would love to see the flight box with the infos about my heading and eta if I'm flying more than 3NM outside of the planned flight path. 

This could be either done by increasing the 3 NM to a bigger number or even better, make it a setting. 

Thank you for the app, it's a already really nice to use.

kebekus, 2025-03-05:
@PhilippBuehler We increased the corridor to 5 NM on your suggestion. The increase appears in v2.32.14, which will go into production on Google Play today. A release for Apple users is in the pipline but will take a while.

We are hesitant to add more settings. The main selling point of Enroute is its simplicity. To keep the app as simple as possible, we try hard to keep the number of setting at an absolute minimum. We hope for your understanding. Please keep reporting any issues that you may find!

closed 2025-03-05 by kebekus

PhilippBuehler, 2025-03-06:
Amazing. That will help me a lot. Thank you for the amazing work

kebekus, 2025-03-07:
It's been my pleasure.