Akaflieg-Freiburg/enroute #506: Importing Trip-Kit from AIP Browser fails

fbbln, 2025-03-09:
**Describe the bug**
A Trip-Kit created for Enroute in AIP Browser results in a .ZIP file.
Importing this to Enroute via "open with" in Total Commander yields an error message about unrecognized Type of file:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Create Trip-Kit by region in AIP Browser
2. Transfer file to phone
3. Open FIle with Enroute
4. See error

**Expected behavior**
I would expect the Trip-Kit to be imported correctly

**Desktop (please complete the following information):**
 - OS: Windows 10

**Smartphone (please complete the following information):**
 - Device: OnePlus Nord 
 - OS: Android 12
 - Total Commander 3.5
 - Enroute 2.32.14

fbbln, 2025-03-09:
I tried two Trip-Kits,
One is complete Germany (and such a Trip-Kit has worked before), and one is North/East.
First one is 128MB, second one is 47MB.
Download link: https://drive.google.com/drive/folders/1vrsU7YpdC3ScZ9Ef8OWFGP2NvQ0NcVaG?usp=sharing

fbbln, 2025-03-11:
@kebekus access was granted. Were you able to download?

kebekus, 2025-03-15:
@fbbln I was able to download the file. On my devices, it imports into Enroute without problems. That leave me wondering about the cause of your problems. Could the file be damaged when you downloaded it with your web browser? Perhaps you could you try one (or all) of the following?

- Download using another web-browser or download program.
- Use a file manager to access your Google drive and copy the file from the drive to local storage.
- Use a file manager to open the file directly from your Google Drive.
- In Enroute, choose 'Import' from the three-dot-menu in the page "Library/Maps and Data", then navigate to the Google Drive and open the file -- WARNING: Upload will take a while, be patient.

Best wishes,

Stefan.

fbbln, 2025-03-15:
Hi @kebekus 
it worked now using the import function within enroute.
I tried both import von local and drive.

I used Open With... in Total Commander, so it seems to be a problem with Total Commander's Open With rather than with the tripkit itself.
I didn't remember that there's in import button for this in Enroute...

Case solved...

closed 2025-03-15 by fbbln

kebekus, 2025-03-15:
@fbbln Thanks for the update. Many happy landings!