Akaflieg-Freiburg/enroute #508: Raster map menu does not appear if only Enroute-provided raster maps installed

lmamane, 2025-03-15:
I downloaded the Swiss raster maps provided by Enroute, and did not import any custom raster map. Now the ICAO Switzerland raster map is shown, but nothing else. The manual says a new button “raster map menu” is supposed to appear, but it does not. I killed the app and restarted it, no change.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to burger menu
2. Click on library / maps and data
3. Scroll down to Europe / Switzerland glider & icao chart
4. download them
5. exit to map

**Expected behavior**
Raster map menu button on main (map) screen

**Screenshots**

[image]

[image]

[image]

 - Device: iPad Pro (11-inch) (4th generation)
 - OS: iPadOS 18.3.1
 - Browser n/a - native app 
 - Version 2.31.16

kebekus, 2025-03-15:
@lmamane Thank you for your report! Can you confirm that the Map-Set ("Aviation Map/Base Map/Terrain Map") are also installed? What version of Enroute do you use?

kebekus, 2025-03-15:
@schneisi Could you have a look? Could it be that the iOS version has fallen behind and does not have the new GUI for raster maps?

schneisi, 2025-03-15:
Yes, unfortunately, I‘ve had some issues lately. 
I expect the latest version to be available early next week.

lmamane, 2025-03-16:
> Can you confirm that the Map-Set ("Aviation Map/Base Map/Terrain Map") are also installed?

Yes, they are. I had a working Enroute (vector) map display before I downloaded the Swiss raster maps.

> What version of Enroute do you use?

 2.31.16

schneisi, 2025-03-20:
Apple approved the new Enroute Version a few moments ago. It should be available to download within the next few hours.

kebekus, 2025-03-22:
@schneisi I just tried 2.32.13 on Anke's iPad and confirm that the issue is solved.

@lmamane Thank you for your report! Please keep reporting any issues that you may find.

closed 2025-03-22 by kebekus

lmamane, 2025-03-22:
Yes, solved in 2.32.13 - GIT #c227aebbc

Thanks