Akaflieg-Freiburg/enroute #509: no notam update anymore

TommoT1, 2025-03-16:
It looks like the notam download does not work anymore....
I'm trying to get notams around Düsseldorf, Germany

 Regards Tom

kebekus, 2025-03-16:
The FAA's NOTAM System is currently experiencing service outages. The FAA reports that technical teams are working to restore the service as soon as possible.

- https://suatest.faa.gov/suatest_banner/suatest_banner.html

Maybe this is a first consequence of Trump/Musk's staff layoff campaign at the FAA.

TommoT1, 2025-03-16:
Hi Stefan,
thanks for the quick response and your great work!
Somehow I expected this kind of answer...
Is there a page where I had been able to find out without filing a bug report?

Regards Tom

kebekus, 2025-03-16:
@TommoT1 Not really. I guess the bug resort is the best way to contact us. I will leave this open as long as the problems persist.

Thanks for the report. Please keep telling us of any issues that you may find.

kebekus, 2025-03-20:
The FAA has restored its service. NOTAM download now works as usual.

@TommoT1 Thank you for the report.

closed 2025-03-20 by kebekus

RobeKrau, 2025-05-06:
Hallo, Stefan @kebekus,

die Störungen beim Abruf von NOTAMs nehmen zu - jedenfalls bei Nutzung von EFN. Bei der Nutzung der App "NOTAM Briefing" hingegen gibt es das Problem nicht. Generell scheinen die Daten hier 
https://notams.aim.faa.gov/notamSearch/nsapp.html#/
abrufbar zu sein.

Wie auch immer ... kann man die Meldung "Die NOTAMs um die aktuelle Poisition sind nicht aktuell, Update wird angefordert" nach einer Weile ausblenden oder kürzen? Sie nimmt gerade auf kleinen Displays und bei Kartenmodus "Flugrichtung oben" viel Platz weg, zumal wenn man eine Luftraum-Höhenbegrenzung eingestellt hat.

Viele Grüße und vielen Dank für Deine unermüdliche Entwicklungsarbeit an EFN!
Robert