Akaflieg-Freiburg/enroute #510: Maps not up-to-date

krheinwald, 2025-03-22:
Map does not show latest changes, i.e. change of VFR mandatory reporting point 'Wedel' to 'Hotel' SW of EDDH

kebekus, 2025-03-22:
Lieber @krheinwald -- Kann es sein, dass sich das Problem inzwischen selbst gelöst hat? Es gab gestern kein Update, weil unsere Infrastruktur zu viele Änderungen sah und deshalb auf manuelle Freigabe wartete. Mit dem Update von heute morgen sollte das Problem aber behoben sein. Kannst Du das bitte noch einmal prüfen und mir kurz rückmelden? -- Danke!

krheinwald, 2025-03-22:
Update wurde jetzt nicht mehr angeboten, nach Neuinstallation der Karte ist jetzt alles OK.

krheinwald, 2025-03-23:
Sorry to reopen the issue: Updated the maps on another device through the notification right now and the map was not updated, i.e. the change of the VFR mandatory reporting point 'Wedel' to 'Hotel' SW of EDDH is not visible.

On the other device, un- and reinstalling the map for Germany fixed it.

Will refrain from updating (un- and reinstalling) to have a device for further tests.

kebekus, 2025-03-24:
I believe there must have been a temporal overlap: the app looks at most once per day for updates. But since I had the release the updates manually, they were not available before noon on that day.

I believe this is normal behavior, and I am therefore closing the issue again. Please feel free to re-open if the problem persists. Thanks for the report!

closed 2025-03-24 by kebekus

krheinwald, 2025-03-26:
@kebekus FYI: Todays map update fixed the problem.