Akaflieg-Freiburg/enroute #511: Installation on Android 12 not possible

Rpal1315, 2025-03-26:
Attempted install on Android 12...Play Store states that it is incompatible with the device. APK is not available for sideloading.Please provide information

charlylima, 2025-04-04:
I have it installed on Android 12 and 14 without problems. Maybe there is an other reason it is incompatible.

It might be a good idea to add some information on the type of smartphone that you use.

kebekus, 2025-04-13:
@Rpal1315 I apologize for the long silence, my day job took the better of me.

As @charlylima points out, Enroute does run on Android 12 devices. Is there any information why the Play Store thinks your device is incompatible? We will definitively need to know more about your device and setup to be able to help.

Developer builds for sideloading are available here:

- https://github.com/Akaflieg-Freiburg/enroute/releases/tag/developerBuilds