Akaflieg-Freiburg/enroute #512: Live Traffic from OGN glidernet.org via internet connection

charlylima, 2025-03-30:
I am writing to let you know that i am currently experimenting on [my branch](https://github.com/charlylima/enroute/blob/ogntraffic/src/traffic/TrafficDataSource_Ogn.h) with adding an other TrafficSource that would download the traffic information from OGN glidernet.org via an internet connection. 

Technically it is very similar to the TrafficDataSource_Tcp,
it will connect to "aprs.glidernet.org" port 14580, 
and then get the data in APRS-IS format which is a text based format similar in principle to NMEA or GDL90. 

```
FLRDF088C>OGFLR,qAS,EDPE:/120958h4852.95N/01111.23E'192/052/A=002464 !W96! id06DF088C -316fpm +4.4rot 22.8dB -19.0kHz gps2x3
FLRDF15A2>OGFLR,qAS,Hilzingen:/120953h4749.74N/00834.14E'218/069/A=005253 !W35! id06DF15A2 -355fpm +0.2rot 7.8dB -11.0kHz gps2x3
ICA4B27B7>OGFLR,qAS,LSPO:/120957h4721.04N\00810.90E^345/106/A=004692 !W95! id214B27B7 -355fpm -0.1rot 13.8dB -11.5kHz gps2x2
ICA4B50A9>OGFLR,qAS,LSPO:/120957h4725.46N/00808.69E'099/080/A=004538 !W92! id054B50A9 -811fpm -2.1rot 3.2dB -0.7kHz gps4x5
FLRDF12DD>OGFLR,qAS,lmr73:/120957h4920.66N/01211.67E'315/048/A=002225 !W76! id06DF12DD +000fpm +2.4rot 8.2dB -5.2kHz gps2x3
ICA3EC1AB>OGFLR,qAS,EDNL:/120957h4800.83N/00954.96E'017/071/A=004564 !W66! id053EC1AB +079fpm -3.9rot 7.5dB -10.3kHz gps2x3
```

Here are some links explaining the network protocol: 
* http://wiki.glidernet.org/wiki:subscribe-to-ogn-data
* http://wiki.glidernet.org/wiki:ogn-flavoured-aprs
* https://github.com/svoop/ogn_client-ruby/wiki/SenderBeacon

Here is the OGN web interface, showing the same data in a webbrowser: 
https://live.glidernet.org

This is similar to feature request #418.

tue-kyndal, 2025-03-30:
It would be awesome if we could add the flarm data to enroute.
Perfekt for glider towing parachute drops 😎

charlylima, 2025-03-30:
[image]
[image]
[image]

tue-kyndal, 2025-03-30:
I will be towing gliders the next 2 days. On Monday and Tuesday. .. Do you have an apk I can sideload... Next to enroute.. Just to check and test.... Or how do I test 😎

tue-kyndal, 2025-03-30:
Will it be possible to add normal transponder data as well... Does flight radar 24 have an api for this?

charlylima, 2025-04-04:
I made good progress this week with the implementation, it should be usable now.
But i was not yet successful to compile it for Android, currently i get no map shown on Android. 

I think there is transponder data included already from the OGN network, for example i could see the airliners starting in EDDM. But currently it would not work as a flight radar 24 application since enroute tries to filter like a anti-collision system to show you only the most relevant aircraft next to you. We can try that later.

kebekus, 2025-04-13:
@charlylima Thanks again for your contribution!

kebekus, 2025-04-13:
@charlylima @tue-kyndal Based on enthusiastic feedback of some of our test-users, we decided that we would like to integrate the feature. We think that the data might be useful to get an overview, but we are absolutely worried that some pilots might use the feature to avoid installing a proper PowerFLARM device. Given the number of collisions that we have in Europe one might think that no sane pilot will ever fly without FLARM, but experience (even at our own airfield) tells us differently.

The challenge will be to come with a design that discourages use of OGN as a sole source of traffic data while not upsetting our users too much. We do not know yet how might best be done.

Random thoughts:

- Experiments by Michael Paus show that sufficient internet connection is available about 40% of flight time, for a typical cross-country flight in Germany.
- My own experiments show that OGN data sometimes comes with a substantial lag.
- Even if a good data connection exists, my own experiments show that OGN data often contains less traffic compared to what my PowerFLARM Fusion sees. We often have rescue helicopters passing in 2.500 ft AGL over my house that do not appear in OGN even though they are FLARM-equipped.
- Our design should definitively not penalize users that have FLARM or ADS-B, but do not have a data connection to their FLARM device; this might apply to paragliders who often have FLARM-out-only devices.
- Our design should not penalize users on the ground.

kebekus, 2025-04-13:
@charlylima I am currently reviewing your code in detail. You can follow the branch ´feature/OGN´ if you want to see what I am doing. Generally speaking, the code looks excellent -- but I might have caught a mix-up with units for ground speed. For that reason, I have started replacing strings by ´Units::Speed´ for data storage.

I will definitively need some time, and I will come back to you on this channel if I have anything to say.

charlylima, 2025-07-08:
@kebekus 
I am happy that you merged it. I have installed the beta on my devices to test it. 

Regarding cleaning up the API and for the unittest, i would like to offer my help, but it is not obvious to me what changes you have in mind. 

If you would like to get rid of the unit test then feel free to do so. Maybe it is better to not spend your valueable time on it. 
I had created the internal class TrafficDataSource_OgnParser that only decodes the protocol and has no other dependencies inside the software, so it could easily be instantiated as a single unit in the unittest. But i got the impression that you did not like it since you already moved code out of it (e.g. formatLoginString()). That will not work well with the unittest without doing the effort to mock lots of other classes, then it might be better to get rid of the test. 

Should we get rid of OgnParser and OgnMessage classes to have the code shorter and move it directly into TrafficDataSource_Ogn or TrafficDataSource_Abstract_APRS ?
Would you like to move everything protocol specific to TrafficDataSource_Abstract_APRS and use TrafficDataSource_Tcp for it?

Whatever you prefer, i would like to offer my help.

VirusPilot, 2025-07-14:
Can't wait to test this new feature, when is this expected to be released for iOS TestFlight? How much would it take to further add a full tracker function so that **enroute** users without any other TX function in their airplane could have the option to report their position to OGN (similar to SafeSky)?

charlylima, 2025-07-14:
@kebekus
That is a good idea to move the OGN Parser to a separate library, similar to metaf library that is used by enroute.
I considered to write a small commandline tool that could replace dump1090 and would use the same library.

By the way, what do you think about referencing enrouteDependencies as a git submodule, just to document which commit of enrouteDependencies should fit to a certain commit of enroute or was used to build the release?

VirusPilot, 2025-07-14:
@charlylima I could not agree more that we need to prevent folks from sending positions while they are driving their car. Maybe an automatic disable when there is no movement could resolve this.

kebekus, 2025-07-14:
@charlylima I have experimented with "enrouteDependencies" as a git submodule, but I gave up after running into a never-ending stream of problems with CMake. I understand that the maplibre people have meanwhile renovated their build system, so maybe this might be the time for a second attempt ... any help is welcome!

VirusPilot, 2025-07-14:
> [@VirusPilot](https://github.com/VirusPilot) You can have both features (OGN data import AND position export) already now by using the third-party app "CCAS VFR-Kollisionswarnsystem", as described [here](https://akaflieg-freiburg.github.io/enrouteManual/04-additionalSoftware/04-trafficInformation.html).

But as I understand, CCAS is **not** providing the position reports back to OGN but only to it's own servers.

VirusPilot, 2025-07-14:
> [@VirusPilot](https://github.com/VirusPilot) I am hesitant about position export to OGN, for several reasons.
> 
> First, I share [@charlylima](https://github.com/charlylima)'s concern about couch pilots appearing on OGN. Perhaps worse, **Enroute Flight Navigation** is also popular within the flight simulation community, and we do not want simulated flights on our screens. Similarly, a certain number of passengers in commercial flights also like to use **Enroute**, and we don't want to duplicate ADS-B data.
> 
> Secondly, and most importantly, I fear that people might use **Enroute Flight Navigation** as a substitute for a proper traffic data receiver. While OGN data can be useful in certain scenarios [flight simulation, ground station at a glider airfield], I strongly recommend against using traffic data from internet services in real flight, for the following reasons.
> 
> * Internet connectivity is not reliable in flight. Even when flying over populated areas in Germany, experience shows that the internet connection fails for about half of the time. This will become worse when the 4G network is phased out, as 5G antennas have a much stronger directional characteristic.
> * Experience in real-world situations shows that data is frequently laggy and often outdated.
> 
> I strongly feel that no responsible pilot should ever fly without a proper traffic data receiver, such as a FLARM or ADS-B device. A PowerFLARM Fusion, the best device on the market, costs 2.210€ at [siebert.aero as of today](https://www.siebert.aero/flugzeug-und-flugplatz/kollisionswarnung-flarm/powerflarm-fusion-flarm-ads-b) (VAT included, plus shipping). This is a tiny fraction of the total cost of an aircraft. There is no excuse for not having this. A proper device saves lifes.

Also here I can't agree more! You don't even have to spend 2k€ as there are many DIY alternatives as well as FLARM latest device PowerFLARM Flex @ 1.500€. I have compiled some info about available devices here: [VirusPilot's Compilation of Collision Avoidance Information](https://gist.github.com/VirusPilot/b6ef1f227123bf80cf491f1c7fce22a4#diy-collision-avoidance-projects-for-general-aviation)

kebekus, 2025-07-15:
@charlylima Done. The repository is now available at [this url](https://github.com/Akaflieg-Freiburg/enrouteOGN), you should have full write access.

kebekus, 2025-07-15:
I am closing this issue now because the feature has been implemented. The iOS port will come in due time.

Thanks to everybody on this channel for your help!