Akaflieg-Freiburg/enroute #513: Traffic Display Inconsistent?

greylingr, 2025-04-03:
**Describe the bug**
=============
First Fantastic App, I use it a lot, and promote it with fellow pilots in South Africa! Kudos!
BUG: It looks like the Stratux traffic is not displayed consistantly if there is no Stratux GPS source, even though there is a DEVICE GPS signal available and in use by Enroute Flight Navigator (EFN) for navigation?

**To Reproduce**
============
I recently started experimenting with Stratux (ADS-B In), and want to build a very small and cheap device to help other pilots build theirs to, to get ADS-B in into their cockpits using your free app too. With a Pi Zero2W now cheap and powerfull it is viable, but it only has 1 USB for the ADSB-Radio and not another for GPS.

My Stratux device (without GPS) is working fine and I can see traffic on other Flight Apps, Like OzRunways and AirnavPro, but it does not show up in EFN?

I get the Traffic Data Receiver GPS error not available on EFN startup, I can see it is communicating fine with EFN in the Settings, Data Connections,  TCP on 192.168.10.1 being lit up Green,
BUT it does not show the traffic I know is there, that is show on other Apps on the Same Device on the Same Stratux Connection.

Can you verify you display traffic, without a Traffic Stratuc GPS? 
If not can you please add the option?
And while you at it can you please add the option to SHOW ALL TRAFFIC, and not only those in close vacinity, since ADS-B out traffic in South Africa is sparse, but definitely increasing. 

**Smartphone (please complete the following information):**
 - Device: Both Android and IOS

charlylima, 2025-04-04:
There is an option in the settings: 
"Primary Position Data Source".
You could try to change that.

greylingr, 2025-04-05:
Hi CharlyLine, thank you for your comments. 

Yes I have seen that option in settings and I have set the "Primary Position Data Source" to use "built-in satnav receiver" of the mobile phone/ipad. There is a valid GPS source on the mobile phone/ipad.

So it does not help and is not at the root of the problem, the problem persists.

I do get a Traffic data receiver warning. "No GPS receiption", which i dismiss because the STRATUX traffic receiver does not have a GPS.  I don't want a GPS on the traffic receiver to lower the cost to build for fellow pilots.

This morning I verified again, no traffic on EFN while traffic is shown on Stratux webpage and OzRunways. So there is a problem on EFN.

Kind Regards
Rudi

kebekus, 2025-04-13:
@greylingr I apologize for the long silence. Let me see if I can help.

- What type of connection do you use to like Enroute with your stratux device? Do you use TCP (with FLARM/NEMA) or UDP (with GDL90)?
- Are you aware that the moving map of Enroute show all the traffic that Enroute knows? To keep the map clean, Enroute shows only traffic that is considered releavant for the pilot. To get a full list of traffic, open the main menu and navigate to "Information/Traffic Data Receiver".

Perhaps it would be helpful if you could post a screenshot of the page "Information/Traffic Data Receiver", taken in a setting where Stratux sees some traffic and Enroute does not.

Best wishes,

Stefan.

kebekus, 2025-04-16:
@greylingr Dear Rudi, 

(1) Connecting the Stratux via TCP seems correct to me. I am a bit unsure about the inner workings of the Stratux device, because the FLARM/NMEA that the Stratux sends give the traffic position relative to the ownship position. I wonder how the Stratux determines this data if it does not have a GPS to determine it's own position. I do not have access a Stratux device, so I can only wonder. Could that be the source of the problem?

(2) Traffic is considered relevant if the vertical distance to the traffic opponent is less than 1,500 m and the horizontal distance is less than 20 NM. Only relevant traffic is shown in the moving map.

I am wondering what your experiments give. Please keep me in the loop!

Best,

Stefan.

kebekus, 2025-05-08:
I am closing this issue for lack of response. Please feel free to re-open it anytime!

closed 2025-05-08 by kebekus

greylingr, 2025-05-08:
Thank you Stefan, did not have an opertunity to fly and experiment and confirm in real aviation conditions yet...

1) I downloaded the latest version of EFN and manage to find the all traffic screen in the latest version. This will help a lot to find discrepencies if they exist.
2) Thank for clarifiying the relevant traffic parameters, I can now use it to verify EFN with another source.

It seems since I fitted a GPS to Stratux the traffic information seems to correlate better, but i need to do more in the air flight testing, since I am in a valley on the ground at home with little overhead traffic at home.

For now let's close this ticket, until i can absolutely proof discrepencies with screenshots.

Thank you

kebekus, 2025-05-08:
@greylingr Very well. Looking forward to hearing back from you.