Akaflieg-Freiburg/enroute #516: Drag & Drop for route creation and manipulation

PilotNubilus, 2025-04-06:
First of all, I really like your App. Especially its reliability due to your focussing on eliminating bugs. I have read the manual and a lot of the other issues and understand the voluntary nature of the project and the manpower limit. But nevertheless, I would like to place some feedback, so that it is out there for possible future consideration.

To my mind creating and especially editing a route is really tedious, because of all the necessary double tapping. It would be so much easier when a new waypoint could be added at the end of the route by holding down on the desired location. As long as it is not released, the waypoint should still be movable for fine tuning. Right now, doing small corrections to a waypoint or changing the route is really annoying, because first of all the old waypoint needs to be deleted by double tapping and selecting the right items in the menu and then the new waypoint needs to be created by more or less the same procedure. It would be such an enhancement, when already existing waypoints could be moved by simply holding down on them and moving them to their new location. Another limitation of the current system that will be overcome is that right now points cannot be placed close together, because the already existing point will always catch the tapping. So routes that go from a point to a special point and then back to the "main route" are not possible.

I have just seen (after writing the above), that there was a post like 5 years ago, where this feature was dismissed, but I kind of don't understand why. Other apps do it too and it is so much better, also during flight. To me it seems way less delicate to firmly press onto the screen and have the chance to place it right, than to double tapp and get what your getting, especially during turbulence.

best regards
Peter

kebekus, 2025-04-13:
@PilotNubilus Thank you for your list of suggestions. Let me try to answer.

- As you point out, improved editing for flight routes has been on our TODO-List for a very long time. The reason for our inactivity is simply lack of manpower. Since our team is small, development is slow and we need to prioritise strictly. We tend prioritse maintenance of our infrastructure and safety-related projects, such as support for traffic receivers.

- We have experimented a bit and found that solutions involving dragging symbols on-screen do not work well in a flight environment, where the pilot has typically only one free hand.

Summary: We are gratefuly for any help that we can get. You will be very welcome if you would like to join the efforts...

TTC70, 2025-05-24:
> First of all, I really like your App. Especially its reliability due to your focussing on eliminating bugs. I have read the manual and a lot of the other issues and understand the voluntary nature [...]

Hello,
I would like to add some suggestions to what Stefan has already stated.
1. to select a point on the map there are two ways: "double tap" or "keep pressed" actions. You are right saying that double tapping is frustrating while in flight, but it is also true that a good flight planning must be do on ground.
2. keeping pressed though can decrease distraction time: it can happen that ATC asks you to go here instead of there, so you have to adapt the flight plan accordingly "on the fly" (pun intended :-), using "keep pressed", instead the frustrating "double tap", is easier also in flight.
3. to add a waypoint to a "closer" waypoint you can proceed as follows: find the GPS coordinates of your point (i.e. Google maps), select any point on the map and save it, go to "modify" and change its coordinates with your waypoint's. There, you have your "close" waypoint saved in your archive, so it can be selected on the flight planning section of the app just typing its name. 
I do this way when i want to add approach waypoints to the specific runaway of an airfield/airport. As you said when the "approach point" is too close to the field and the app doesn't allow you to select it. With this "trick" you can.

Cheers