Akaflieg-Freiburg/enroute #517: App startet nicht

wo-wi, 2025-04-20:
Die App starte nicht mehr.
Es müssen alle Anwendungen zuvor geschlossen werden. Danach startet die app

**Smartphone (please complete the following information):**
 - Device: samsung a50
 - OS: Android 11

wolfkass, 2025-04-28:
Die App sartet nicht mehr. Nach einer Neuinstallation kann man es wieder öffnen aber auch nur 2-3 mal

Device  Redmi 9a
OS Android 10

kebekus, 2025-05-06:
Danke für die Meldungen. Das ist für mich ein schwierig zu lösendes Problem, weil der Fehler auf meinen eigenen Geräten nicht auftritt. Könnt ihr mir bitte mit folgenden Daten helfen?

- Tritt der Fehler auch mit der aktuellen Version 2.33.2 auf [die vor wenigen Stunden bei Google Play erschienen ist]?
- Was genau passiert, wenn Ihr die App startet?
- Habt ihr bei Euch (oder bei Familie/Fliegerkollegen) noch ein Zweitgerät, auf dem ihr Enroute Flight Navigation ausprobieren könnt. Es wäre interessant zu wissen, ob unsere App bei Euch auf anderen Geräten läuft.
- Ein mögliches Problem ist Speichermangel. Könnt Ihr bitte einmal schauen, wieviel Speicher im Gerät noch frei ist?
- Ein mögliches Problem ist unzuverlässiger oder kaputter Speicher. Läuft das System ansonsten zuverlässig? Laufen andere Apps ähnlicher Größe zuverlässig?

Beste Grüße aus EDTF,

Stefan.

wo-wi, 2025-05-07:
Mit dm neusten Update (7.5.2025) ist das Problem gelöst. App startet ohne Probleme.

closed 2025-05-07 by kebekus

wolfkass, 2025-05-10:
Hallo, ich habe die letzten Tage das Programm immer wieder mal geöffnet und anscheinend ist der Fehler behoben. Vielen Dank für eure Mühe. Wolfgang 

Closed #517 as completed.

kebekus, 2025-05-10:
@wolfkass Danke für die Nachricht. Wir freuen uns (und sind ein bischen erleichert), dass jetzt alles läuft.