Akaflieg-Freiburg/enroute #519: Export Library for Sync on another Device

fbbln, 2025-04-30:
**Is your feature request related to a problem? Please describe.**
Currently, if I use Enroute on multiple devices I have to set up each one individually in regards to airplanes and routes.
- Routes can be individually exported but not all at once
- Airplanes can be setup but not exported at all
Importing is also tedious as I need to import each route individually.

**Describe the solution you'd like**
Minimum solution would be to be able to export airplanes and to bulk export routes.
Same for importing: bulk importing a complete folder

Optimal solution would be to have a Sync option which copies all local routes/airplanes/waypoints to a shared folder (e.g. Google Drive or similar) either if something has changed or via a Sync-Button on demand.
Same goes for the import from a shared folder.

I know that a real sync creates issues (what dataset takes precedence etc.) but with a "manual" sync this can be overcome if one has to choose sync out or sync in.

kebekus, 2025-05-06:
@fbbln Thanks for the suggestion. This is a very useful feature that we should definitively implement. HOWEVER ... for lack of development ressources and manpower I cannot promise that we can do it quickly. Any help is greatly welcome!

HansBorjes, 2025-07-24:
For an easy solution I would suggest to implement storage of all data in the related cloud profile of the platform.

Apple and Google both offer to store all app related data using their cloud services. If this would be an option inside Enroute app it might be easy to implement. OS would take care to buffer locally without internet connection and sync once available.

As Enroute works on MacOS with Apple Silicon and iPhone this would be a seamless combo.

Same for ChromeBook and Android smartphones.

kebekus, 2025-12-23:
@fbbln @HansBorjes Your push has been received. However, I cannot promise that the feature will come quickly. We are a small team only, and already spend a lot of time keeping the infrastructure going. Our requests for help were largely unanswered. Unless more people join in, development is slow and we need to prioritise strictly.

Please let us know if you want to help with the development. Implementing this feature could be a great start!