Akaflieg-Freiburg/enroute #521: Custom base maps rendering issue 2.0

PilotNubilus, 2025-05-06:
Since the chairman of my flying club suggested your app to me, I love it. He told me, that he even had a short chat with you on the phone and you were very nice.

**Describe the bug**
I have the same problem as in post #238. I am sorry if I should not have opened a new issue, but I am not sure whether writing in a closed one is a good idea. I really like the app overall, but one of the very few downsides is, that I think the maps could have a bit more details. Where I fly (Austria) even fairly large villages are not drawn on the map. I found a different post, where you said, that local enhancements are not possible, because there can only be one grade of detail for all the maps. I understand that, but I think a bit more details would be great and the increase in file size of the already very small maps is maybe not that big of a deal? Due to the official nature and relevant information I plan my flights in the ICAO map. Because a lot of features of said map are not included in the Enroute map it is sometimes a bit tedious to find the planed points in Enroute (using Satellitenansicht). Hence overlaying the ICAO map solves this planning deficiency nicely. And for flying it can be hid again, because vector is just superior. I am sorry for the long story. But despite having ok-ish quality (visible when imported back in QGIS), in Enroute the quality of the georeferenced mbtiles raster ICAO map (created in QGIS) is just terrible.

**To Reproduce**
Steps to reproduce the behavior:
1. Download official ICAO map from austro control (pdf)
2. georeference it in QGIS 3.40.3 and save it as mbtiles file
3. import it into Enroute via menu / Bibliothek / Karten und Daten / three dots / Importieren...
4. make it visible in the layers menu on the map

**Expected behavior**
Seeing the ICAO map overlay with a resolution close to the of the imported mbtiles file.

**Screenshots**
[image]

**Smartphone:**
 - Device: Huawei Mate 10 Pro
 - OS: EMUI 12.0.0
 - Enroute Version: v2.33.2

**Additional context**
NIL

kebekus, 2025-05-07:
@PilotNubilus Thanks for your report. Can you please upload all relevant files so I can look at them?

- https://nextcloud.cplx.vm.uni-freiburg.de/index.php/s/M6WBjZAt88qc2qT

Please let me know once you are done, so I can close the upload link.

PilotNubilus, 2025-05-07:
I have uploaded the .mbtiles and .tif map file. I am not entirely sure whether it worked, please have a look.

The ICAO map I used can be found here:

[ICAO map Austria](https://www.austrocontrol.at/icao_karte)

best regards

kebekus, 2025-05-07:
@PilotNubilus The files have arrived, thanks. I shall look at them and come back to you. Please give me a few days.

kebekus, 2025-05-07:
@PilotNubilus As far as I can tell, the problem lies with your MBTILES file and not with Enroute Flight Navigation. The metadata of your files claims to support zoom levels 0-12, but tiles exist only for zoom levels 8, 9, 10, 11, and 12. The raster data is fairly uncompressed PNG, which makes your file extremely large.

For comparison, I have run your GeoTIFF through my own scripts and produced an MBTILES file of about 20% the size of yours, which contains tiles for all relevant zoom levels, uses state-of-the-art WEBP compression and gives better-looking results. 

- https://nextcloud.cplx.vm.uni-freiburg.de/index.php/s/7Be2EPZq2C7eRjZ

Please let me know once you are done, so I can close the download link.

PilotNubilus, 2025-05-07:
Thank you for checking. I have downloaded your file and it works.

Because you used my .tif and it worked I assume that the georeference process to get the .tif works. To get the .mbtiles file I "simply" export it in QGIS as stated in the Enroute manual. See screenshot. Why is it not working then? Can someone help, I would really like that to work.

[image]

best regards

kebekus, 2025-05-08:
@PilotNubilus For a long-term solution, I recommend lobbying at Austrocontrol. If Austrocontrol follows the Swiss example to publish GeoTIFF under license conditions that allow inclusion in Enroute, then I can update our infrastructure to generate raster maps automatically. Users can then use the maps without any hassle at all, and receive automatic updates.

kebekus, 2025-05-08:
I am closing this issue now because I do not see any actions I could take. 

@PilotNubilus Thanks for your report. Please keep reporting any issues that you find!

closed 2025-05-08 by kebekus

PilotNubilus, 2025-05-08:
@kebekus Which software do you use to get the MBTILES from the GeoTIFF? Or can you suggest something I can use?

I added writing a post in the QGIS forum and writing an e-mail to Austrocontrol to my todo list.

best regards

kebekus, 2025-05-08:
@PilotNubilus I did not find anything that worked sufficiently well, so I programmed a little script myself. You find my scripts here:

- https://github.com/Akaflieg-Freiburg/enrouteServer

PilotNubilus, 2025-05-12:
@kebekus thank you for writing, publishing and informing me about your python scripts. I was already able to convert from GeoTIFF to MBTILES with it.

As said above, I made a [post](https://gis.stackexchange.com/questions/492379/converting-mbtiles-from-tif-with-qgis-does-not-create-all-zoom-levels) in the GIS forum.

Concerning getting in touch with Austrocontrol, I got hold of the right person and would like to forward their response to you, but I don`t want to paste the e-mail in here due to data protection. How can I send it to you privately?

best regards

kebekus, 2025-05-13:
@PilotNubilus Find my e-mail address in the app, under Main Menu/Information/About Enroute/Authors.

PilotNubilus, 2025-05-13:
@kebekus thank you, I have already sent it to you.

While checking out the MBTILES file you provided via the link above, I noticed something. When zooming out in Enroute at some point it gets suddenly noticeably blurrier, whereas it stays sharp all the way when I use your file in a different navigation app (no sudden change in blurriness). Is this expected behaviour withing the cababilities of Enroute, or something to look into?

[image]

[image]

best regards