Akaflieg-Freiburg/enroute #522: .CUP files from Austro Control are not supported

PilotNubilus, 2025-05-09:
The .CUP waypoint files provided by Austro Control [here](https://www.austrocontrol.at/piloten/vor_dem_flug/aim_produkte/luftraumstruktur) are not supported. It seems to me that Enroute does not support the latest .CUP file format specified [here](https://github.com/naviter/seeyou_file_formats/releases/latest/download/CUP_file_format.pdf). The .CUP file from Austro Control has the two columns userdata and pics. When they and the corresponding commas in every row are deleted, it can be imported without any problems. I personally do not need it, and even if so, the file could easily be adjusted in a minute with any text editor and it's search and replace function, but I was not sure whether this is intentional, or an unkown issue.

best regards

kebekus, 2025-05-10:
@PilotNubilus I can confirm the issue. Enroute can read CUP files conforming to the 2018 specification, but does not support the 2024 specification. This should be straightforward to implement, so I hope to have an update soon.

kebekus, 2025-05-11:
Fixed in 2.33.3, which appeared on Google Play today.

@PilotNubilus Thanks for reporting the issue! Without help from our users, we would not be able to find (and then fix) all issues. Feel free to re-open this issue if the problem persists.

closed 2025-05-11 by kebekus