Akaflieg-Freiburg/enroute #523: Double tap gesture does not work without any waypoints installed

PilotNubilus, 2025-05-09:
**Describe the bug**
I tried using Enroute only with personal map and airspace data. But the double tap gesture on the map only works if there is at least one waypoint already in the library.

**To Reproduce**
Steps to reproduce the behavior:
1. uninstall all maps downloaded via Enroute
2. delete all waypoints
3. import a custom raster map
4. import custom airspace data
5. double tap on the map

**Expected behavior**
After double tapping the waypoint dialog should open. When there is saved at least one waypoint before deleting the enroute map, double tapping still works after installing a custom raster map and a custom airspace file.

**Alternate Way to Reproduce**
1. have at least one waypoint installed
2. uninstall all maps downloaded via Enroute
3. import a custom raster map
4. import custom airspace data
5. check that double tap works
6. delete all waypoints
7. check that double tap does not work any more

**Additional context**
Enroute version v2.33.2

best regards

kebekus, 2025-05-10:
I can confirm this behaviour. To be honest, Enroute was never designed to work without maps installed, so this has never been tested. This issue should hopefully be straightforward to fix.

closed 2025-05-10 by kebekus

PilotNubilus, 2025-05-10:
@kebekus that was quick, thank you!

kebekus, 2025-05-11:
Fixed in 2.33.3, which appeared on Google Play today.

@PilotNubilus Thanks for reporting the issue! Without help from our users, we would not be able to find (and then fix) all issues. Feel free to re-open this issue if the problem persists.

commit 892c1d52 (2025-05-10): Fix #523