Akaflieg-Freiburg/enroute #533: graphic error

flyinkevin, 2025-07-17:
When opening the app on my tablet Lenovo Tab M8 (model TB301XU) it only shows a White Screen. This First Bug I also Had on my Xiaomi Smartphone. But while my Smartphone after a few restarts and several tries managed to Work Out, my tablet Just starts flickering, shows a broken down graphic and does Not recover anymore.

Even after a restart, installing all actual Updates etc it does Not come Back to a normal Screen. I think its a graphical Bug, because when touching and Holding a Finger longer on the wild flickering Screen you can See a window opening...so the app IS working in the Background but IT isnt able to Show the right Screen:

[image]
[image]
[image]

charlylima, 2025-07-21:
I had the same problem on one of my tablets (Ulefone Armor Pad).
I got the impression that it could sporadically get into this error state when the app was killed on the first use of Enroute. Then it would not start again. Maybe it could also happen after an update of the app. 

I could work around it by deleting the cache of the app:
**Android Settings -> Apps -> Enroute Flight Navigation -> Storage and cache -> Clear Cache.**
Afterwards i started Enroute successfully and also did a clean exit by selecting Exit from the Menu in Enroute, not sure if this clean exit is necessary. Since then I never had the issue again. 

This feels for me like it is probably an issue of Qt on Android, not of the application itself.

kebekus, 2025-08-06:
@flyinkevin Thanks for your report. I am closing this issue now. Please feel free to reopen if new problems appear!

closed 2025-08-06 by kebekus