Akaflieg-Freiburg/enroute #535: MacOS Crash on route export

HansBorjes, 2025-07-23:
Route export crashes application on MacBook. Works on iPhone.

 - Device: MacBook Air M2
 - OS: MacOS 15.5

schneisi, 2025-07-27:
I can reproduce this issue and will try to fix it in the coming days.

charlylima, 2025-07-29:
For the route export, I created the new function "PlatformAdaptor_Abstract::setClipboardText(..)". 
Maybe the problem is in this function, this is platform specific code:
https://github.com/Akaflieg-Freiburg/enroute/blob/424241e52e68969a321a81543574e1e49594a09e/src/platform/PlatformAdaptor_Abstract.cpp#L52-L59

It is called from QML: 
https://github.com/Akaflieg-Freiburg/enroute/blob/424241e52e68969a321a81543574e1e49594a09e/src/qml/pages/FlightRouteEditor.qml#L399-L415

charlylima, 2025-07-29:
Could you try if this commit fixes the issue on MacBook?
https://github.com/charlylima/enroute/commit/5a0e4d6e8f9082c4507acbf8b7ee6fa077f6d955

HansBorjes, 2025-08-03:
I would be willing to test, but can only test binaries. I don't have a build toolchain.

schneisi, 2025-08-19:
Unfortunately, this didn't fix the crash.
It crashes inside iOS/macOS internals I have no idea about right now.

I have to dig deeper here...

kebekus, 2025-11-06:
@HansBorjes We looked into the issue. It seems that the problems are triggered by differences between original iOS devices and the MacOS subsystem that runs iOS apps on your Mac.

As a solution, we are currently working on a version of "Enroute" built specifically for Mac. Progress will be reported on this. chanel, and we will soon be looking for beta-testers.

kebekus, 2025-11-08:
@HansBorjes @charlylima To prepare for the upcoming version of "Enroute for macOS", we have ported Enroute's little brother, the app "Add Times" to macOS. If you can find the time, we would appreciate it if you could beta-test "Add Times for macOS" and tell us about your experience. Joint the beta-test programm on "Apple Testflight" via the link below:

- https://testflight.apple.com/join/PITm0Dqj

kebekus, 2025-12-14:
Fixed in 3.0.10, which is ready to be distributed to macOS. Simon Schneider will coordinate the release and the fight with the Apple burocracy. I leave this issue report open until the macOS version appears on the App Store.

kebekus, 2026-10-17:
@HansBorjes Good news, and thank you for your patience! **Enroute for macOS** is now available on the App Store:

https://apps.apple.com/de/app/enroute-flight-navigation/id6448892176

This is a native version built specifically for the Mac, not the iOS app running in Apple's compatibility layer. That layer caused the crash you reported, and route export now works on the Mac without crashing.

Thank you for reporting the problem and for offering to test. Thanks also to @charlylima and @schneisi for tracking it down. Your reports helped push us to build a proper Mac version.

If anything doesn't work as expected in the new macOS app, please open a new issue. I'm closing this one now. Happy flying! ✈️

closed 2026-10-17 by kebekus