Akaflieg-Freiburg/enroute #537: XCNav serial connections not available

yorickreum, 2025-07-29:
**Describe the bug**
On our rk3566/Android-based XCNav, serial connections are greyed out. XCSoar on the same device can connect to the XCVario via serial connection without any issues. 

**To Reproduce**
Steps to reproduce the behavior:
Open Enroute on an XCNav with serial devices connected. Go to the connect data sources tab and try to connect the serial devices. The option will be grayed out.

**Expected behavior**
Being able to select the serial devices.

**Smartphone (please complete the following information):**
 - Device: XCNav
 - OS: Android Tablet
 - Version: 2.34.2

kebekus, 2025-07-30:
@yorickreum This is on our TODO List, see #532. We will probably start working on this in fall, once the "sideview" feature is implemented. Our team is presently working at the limit, and any help is greatly welcome!

kebekus, 2025-07-30:
@yorickreum Wo fliegst Du eigentlich?

kebekus, 2025-12-14:
@yorickreum @XCNav I apologize for the long silence. I intend to work on this issue next, so *maybe* we have a solution next year.

kebekus, 2025-12-27:
@XCNav As far as I can see, Android typically uses Unix file permissions to disallow access to device files such as /dev/ttyS0. How is that solved in XCNav? Have you modifies Android somehow to allow non-root access to those files? Have you modified the file permissions? Or does the user process run with root previleges?

@XCNav @hjr Would it be possible to borrow a device for developing and testing purposes?

XCNav, 2025-12-27:
No, we use stock android. Permission handling is all done by XCSoar or LK8000. On older versions of XCNAV the serial input is handled via FT232 to the internal USB ports, the newer version handles it via the on-board serial ttl ports. We. Could borrow some barebone mainboard with the connector board wich could be connected via HDMI to a screen, if that would help.

kebekus, 2025-12-27:
@XCNav Thanks for your prompt reply. I would greatly appreciate the testing device that you describe, as I am currently stuck. My address: 

- Stefan Kebekus, Wintererstr. 77, 79104 Freiburg, GERMANY

Side questions: HDMI output is fine with me, but how do I input? With a connected mouse and keyboard, I presume? I understand that I can set the device to "developer mode", which allows installing development versions of my app via the Android Developer Bridge (=adb) and USB?

XCNav, 2026-01-03:
Very nice, I am still abroad on vacation, will test when back home.

kebekus, 2026-01-18:
Dear all, I believe the feature is implemented in Version 3.1.0, which appeared in production on Google Play a few minutes ago. For lack of a testing device, I was unable to try the feature on an XCNav device, but I can confirm that it works on the "XC Touch Nav".

I will close this issue now, but I would still value any feedback. Please let me know if Version 3.1.0 works for you!

Thank you for your help.

Stefan.

PS: Even though the issue is closed, replying here will still reach me.