Akaflieg-Freiburg/enroute #543: iPhone gets hot

HansBorjes, 2025-08-31:
**Describe the bug**
My iPhone gets really hot when running Enroute

 - iOS 26.0 (23A5330a), but was the same with 18.6.2

iPhone 16e

WiFi @home

schneisi, 2025-09-15:
Interesting observations. I will take a closer look, but this may take a couple of days due to my personal schedule

kebekus, 2026-10-31:
@HansBorjes This issue report has been lying idle for a while; I apologize. There have been major updates in the underlying technology stack since the last activity on this thread, which might already have fixed the issue.

Has the problem been resolved in one of the more current versions of Enroute, or does it still persist?

schneisi, 2026-11-15:
v3.5.16 will include improvements when using OGN.
I expect a jump in performance when we can finally switch to Metal with Qt-Maplibre Version 4, but we don't know when v4 will be released.