Akaflieg-Freiburg/enroute #545: Additional layer ifr reporting points

kebekus, 2025-08-31:
Sirs,
I am enthusiast user of enroute. From the north east Italy, last year I flown to the Atlantic coast an this year to the Baltic sea. As I am flying with an average speed of 65kt, seldomly I can report every 30 minutes over a vfr reporting point. 
Futhermore  often I need to file a flight plan around (or below) TMZ and or TMA, where no VFR reporting points are set.
For those reasons, I do find very useful to state intentions and to report the IFR reporting points.
They are published and updated at:
https://www.eurocontrol.int/publication/free-route-airspace-fra-points-list-ecac-area

How can I contribute to:
1 Ad a button on enroute to visualise o hidden an additional layer with the IFR reporting points?
2 set an automatic check for updates in the above indicated dBASE?

Thank you very much for the work done up today.

Happy landings!

Giuseppe Cane 🙂

fbbln, 2025-08-31:
+1
I also had the experience for example in Italy and Croatia where traffic controlers like to send you from one IFR point to the next. Showing them could be of great help then.

TTC70, 2025-09-01:
Hey Stefan,
the request is a bit strange.
As far as I know, as VFR/ULM pilot, I can say that there are no visual reporting points in AirSpace class G.
It is also true that the communication to the FIC (Flight Information Center) is not requested if you do not file a flight plan with Italian ARO and Eurocontrol.
And if you file a flight plan the only radio obligation is to report "normal operation" every 30 minutes. For this reasons (and many other) there is no need to have a specific VRP.
A list of IFR waypoint (which btw I have my doubts that they will be "managed" with happiness by the ATC Controllers) could be a "heavy" add.on to Enroute..
On the other side, also a mountain, a monaster, a bay, and so on, could be a Visual reporting point, known by the local ATC.
If a controller is asking to turn toward a IFR reporting point, it is quite strange, and the pilot can ask to refer to a proper VRP.
...but this is just my idea.

DimoDKENV, 2025-09-20:
Thanks for considering the issue, as well for the comments.
Flying in only G space, it is quite different from a cross country flying. A cross country flying means also flying among b, c, d, e and g spaces. Flying in a g space below a c space, get the FIC rather nervous. Please go fly over Schaffhausen and listen to Zürich, how they get nervous if you do not provide a sound knowledge of the airspace. For example, last month I was getting back from LKCH to LIKH. After EDME, Langen ask me if it was my intention to cross both controlled airspaces of München and Salzburg. I told Langen it was my intention to fly via GOMIG,TITIG,,UNKEN. Langen die Not rise any comment.
I cannot be familiar wir some conspicuous points or landmarks of the territory  (Castles, bridges, crossroads ...) I am crossing, so it can be very tricky flying and being able to report something I am not familiar to. The ifr reporting points are an easy solution for someone who doesn't know any inches of the flying territory . Last fact, two weeks ago Padova Radar 119.175 told me to report LUPIN. I was flying VFR...
Thank you Professor Kebekus!
Happy landings! Giuseppe Cane 🙂