Akaflieg-Freiburg/enroute #547: App freeze iOS

gallmeier5010-LS4, 2025-09-10:
**Describe the bug**
When you select a red exclamation mark (e. g. NOTAM), the app freezes. The app needs to be completely closed and reopened.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Enroute
2. Selecting red exclamation mark

**Expected behavior**
The information has to open, the app has to go on

**Desktop**
- OS: iOS: 18.6.2
 - Version Enroute 2.32.14

**Smartphone (please complete the following information):**
 - Device: iPhone 12 Pro
 - OS: iOS 18.6.2
 - Version Enroute 2.32.14

gallmeier5010-LS4, 2025-09-22:
App freezes also when another operation is chosen

schneisi, 2025-09-25:
I am unable to reproduce this issue. 
Enroute 2.34 2.34 will be available on the App Store within the next few hours.
Please try again after installing the update.

closed 2025-11-02 by schneisi