Akaflieg-Freiburg/enroute #550: Mismatch between altitude AMSL and AGL when flying on sea.

TTC70, 2025-09-15:
**Describe the bug**
The title says it all. When flying on sea and you set AGL (tapping on the left side bottom screen) the reading is off.
Maybe it reads the depth of the sea bottom?!

**To Reproduce**
Steps to reproduce the behavior:
1. Fly on the seas
2. Switch between AGL and AMSL altitude
3. Read the mismatch

**Expected behavior**
When flying on the sea the reading AMSL and AGL should show the same number. It's understood that "Above Ground Level" has more sense when flying on ground, but when flying on the sea the two readings should be the same.

**Screenshots**
n.a..

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S10
 - OS: Android 12
 - Browser n.a.
 - Version n.a.

**Additional context**
n.a.

kebekus, 2025-09-16:
I fear that indeed, Enroute takes the sea bottom as "ground". This is of course very wrong. A bugfix is unfortunately not easy, and will therefore take its time.

TTC70, 2025-11-11:
Hey Stefan,

as update to the previous issue, i would like to highlight another problem which is common to this issue and to the "lateral view" issues.

As you can see from the screenshot, the airspace "lower than the plane - GND-2500ft" falls in the sea and its height is not correctly shown.

Furthermore, there are same color aberrations in the representation of the upper airspace.

[image]

Cheers.

Antonio

kebekus, 2025-11-13:
Update: This issue is the next in my TODO-List. I hope to have a fix soon.

TTC70, 2025-11-14:
> [@TTC70](https://github.com/TTC70) Fixed, hopefully. If you update your maps now, there should be a big update in the terrain maps. Afterwards, the problem should be fixed. Could you please try this out and report back? -- Thanks!

Wilco.

kebekus, 2025-11-16:
@TTC70 Thanks for reporting the issue.